langemand/SoulEVSpy #1: Many errors in SOH calculation.

JejuSoul, 2019-02-16:
1. You write
 // Total battery capacity (totcap) for the 2015-2017 models were 30.5 kWh, for the 2018 onwards 31.8 kWh
 // according to Wikipedia: https://en.wikipedia.org/wiki/Kia_Soul_EV

The Wikipedia article is almost certainly wrong about the 30kWh Soul EV total capacity.
A better guess, based on the 27kWh SoulEV having 30.5kWh total capacity would be 33.5kWh.
Until someone takes the battery out and does a lab test we don't know for sure.

2. The 30kWh Soul EV does not have values for min and max deterioration. So your attempted calculation is pointless anyway.

3. Your formula is also incorrect.
 double sohpct = (totcap * (1-detmax/100.0) / nomcap * 100.0);

The function should use the average of min and max deterioration and subtract from 110%
This will then match the value as given by the official Kia GDS. (If >100 SOH is shown as 100% at Kia).
The SOH we want is the remaining usable capacity. That is what Kia uses for determining battery warranties.

langemand, 2019-02-16:
Hello Jeju, 
thank you for the corrections.

Do I understand it correctly: 
The 2018 Kia Soul EV does not have an SOH value on the printout, when checked by the Kia garage (official GDS)?
I don’t know what “the official Kia GDS” means.

Regarding the min and max det, I guess this is wrong too, then: 
https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/blob/master/Soul%20EV%20-%2030kWh/extendedpids/Kia_Soul_EV_BMS_data.csv

Regarding the 2015 27 kWh version:
The relevant “original total” battery capacity is then 29.7 kWh, not 30.5 kWh, if a SOH of 100% refers to, I guess.

I have corrected the SOH calculation in SoulEVSpy according to your suggestions, and made the new version available for download:
http://spjeldager.dk/elbil/SoulEVSpy-0.1.2-2619-debug.apk 
It will display “unknown deterioration” for the 30 kWh version, and up to 110% for the 27 kWh version.

Thanks again!
Henrik
-- 

Venlig hilsen / best regards 
Henrik R. Scheel

Spjeldager Consult ApS
Spjeldager 9 
DK-2630 Taastrup 
CVR: 34491399 
T: +45 2720 9828 
http://www.spjeldager.dk/

JejuSoul, 2019-02-18:
GDS / KDS is explained here - http://www.mykiasoulev.com/forum/viewtopic.php?f=6&t=434&p=4340#p4340

Yes the codes on my site for the 30kWh Soul EV Min and Max deterioration are wrong too.
I haven't corrected them because we don't know hat they show yet.
Discussion is here - http://www.mykiasoulev.com/forum/viewtopic.php?f=6&t=471&p=8116&hilit=30kWh+max+det#p8116

I will test the new version today

closed 2019-02-23 by langemand