langemand/SoulEVSpy #12: SoulEVSpy not working on the Ioniq Electric

JejuSoul, 2019-03-18:
I asked Ioniq users to test the app.
One user has tested the app.
He provides logs and a YouTube video. It shows no data being received.
The BMS on the Ioniq is identical to ours so it should be showing something!

https://www.ioniqforum.com/forum/395624-post51.html

JejuSoul, 2019-03-18:
The app EVNotify has a lot of connection data for both the Soul EV and the Ioniq Electric.
See - https://github.com/EVNotify/EVNotify/tree/2448556ac307f90941d8eeb9bb7a5c5933761c5e/app/www/components/cars

This app is much more complex than when I first tested it a  few years ago, I am going to take a look at it again

langemand, 2019-04-02:
Thanks for the tip! I have modified the initialization sequence to closer resemble the sequence in EVNotify (in 0.1.3-2999), and at least one user was able to successfully see the BMS data from Ionic, although updating at 21 second intervals, due to the many timeouts for Soul-data that isn't there.

langemand, 2019-07-23:
I have received recorded files from a Ioniq user (BETA version 0.1.0-3853), scanning BMS-data every 3rd second, without any issues.

closed 2019-07-23 by langemand