langemand/SoulEVSpy #32: VIN not decoded for a Kona.

EricYoungdale, 2019-12-24:
My VIN starts with:

KM8K53AG5KU

The code in KiaVinParser.java doesn't handle the case of wmi.equals("KM8")

langemand, 2019-12-24:
Thanks, I will add that combination. Merry Christmas! --Best regards / venlig hilsenHenrik ScheelMobil: +45 2720 9828-------- Oprindelig meddelelse --------Emne: [langemand/SoulEVSpy] VIN not decoded for a Kona. (#32)Fra: EricYoungdale Til: langemand/SoulEVSpy CC: Subscribed My VIN starts with:
KM8K53AG5KU
The code in KiaVinParser.java doesn't handle the case of wmi.equals("KM8")

langemand, 2019-12-25:
Hello again.
I am updating the KiaVinParser code.

The 5th character in the VIN identifies the car trim level. Yours is ‘5', which is new to me. What is the name of the trim level for your car (other trim levels are Trend, Exclusive, SE, Base, …)?

I wish you a Merry Christmas.

Best regards,
Henrik
-- 

Venlig hilsen / best regards 
Henrik R. Scheel

Spjeldager Consult ApS
Spjeldager 9 
DK-2630 Taastrup 
CVR: 34491399 
T: +45 2720 9828 
http://www.spjeldager.dk/ <http://www.spjeldager.dk/>

EricYoungdale, 2019-12-25:
> Hello again. I am updating the KiaVinParser code. The 5th character in the VIN identifies the car trim level. Yours is ‘5', which is new to me. What is the name of the trim level for your car (other [...]

It is the "Ultimate" trim.  That's what they call it in the States, anyways

EricYoungdale, 2019-12-25:
One other thing that might be relevant.  The Kona in the US does not have a heat pump (PTC heater only).  And there is no battery heater either.  The Canadian versions of the car have both of these things. 

In 2020, they added the battery heater to US models, so it is only the 2019 models that have this quirk.  I don't recall whether all trim levels have this.

The BMS page shows funny numbers for the temperature of the non-existent battery heater.  I had not yet gotten around to collecting the logs and creating a 2nd issue.

langemand, 2019-12-28:
Hi again.
I don’t know if you are using the paid version (Soul EV Spy) or the free version (Soul EV Spy Lite); but I have released a fix for this in the latest BETA-release of the paid version: 1.0.0-340.

Thanks for informing me of the bug!

Best regards,
Henrik
-- 

Venlig hilsen / best regards 
Henrik R. Scheel

Spjeldager Consult ApS
Spjeldager 9 
DK-2630 Taastrup 
CVR: 34491399 
T: +45 2720 9828 
http://www.spjeldager.dk/

closed 2020-09-26 by langemand